a2633063/zM1 #2: 刷机成功但是连不上wifi

walk728, 2019-11-24:
连接zm1的热点以后，可以打开192.168.0.1，填入wifi名称密码，现成配置成功，但是设备就是连接不上wifi，测试其他设备均可以连上填入的2.4gwifi。重置再连一样连接不上

walk728, 2019-11-24:
我换了个ap就好了，看来是加密协议的问题。

jokerlala, 2019-12-26:
刷机成功，现需要接入其他的网络，该如何重置呢？

a2633063, 2020-01-01:
> 刷机成功，现需要接入其他的网络，该如何重置呢？

长按按键直到屏幕所有内容点亮后松开(大概15-20秒) 即可重新配网

maxmib, 2020-01-01:
@a2633063 接入无线后，直接访问设备ip是不是也有web界面配置无线，就是app里一直添加不了局域网设备发现不了m1，怎么确定是固件没刷好还是网络原因

jokerlala, 2020-01-02:
好的，非常感谢您的回复

ciya99, 2020-03-17:
> 连接zm1的热点以后，可以打开192.168.0.1，填入wifi名称密码，现成配置成功，但是设备就是连接不上wifi，测试其他设备均可以连上填入的2.4gwifi。重置再连一样连接不上

我和你同样的问题，设置好后无法连接WIFI，（K2T集客固件）BTW:tc1连接一切正常。奇怪这是什么原因。

walk728, 2020-03-17:
> > 连接zm1的热点以后，可以打开192.168.0.1，填入wifi名称密码，现成配置成功，但是设备就是连接不上wifi，测试其他设备均可以连上填入的2.4gwifi。重置再连一样连接不上
> 
> 我和你同样的问题，设置好后无法连接WIFI，（K2T集客固件）BTW:tc1连接一切正常。奇怪这是什么原因。

我的正好也是K2T集客固件！

ciya99, 2020-03-18:
> > > 连接zm1的热点以后，可以打开192.168.0.1，填入wifi名称密码，现成配置成功，但是设备就是连接不上wifi，测试其他设备均可以连上填入的2.4gwifi。重置再连一样连接不上
> > 
> > 
> > 我和你同样的问题，设置好后无法连接WIFI，（K2T集客固件）BTW:tc1连接一切正常。奇怪这是什么原因。
> 
> 我的正好也是K2T集客固件！

嗯， 更换成K3当AP就正常了，K2T反复改了几个无线配置，还是连不上。

smartsilver, 2020-04-02:
> > > > 连接zm1的热点以后，可以打开192.168.0.1，填入wifi名称密码，现成配置成功，但是设备就是连接不上wifi，测试其他设备均可以连上填入的2.4gwifi。重置再连一样连接不上
> > > 
> > > 
> > > 我和你同样的问题，设置好后无法连接WIFI，（K2T集客固件）BTW:tc1连接一切正常。奇怪这是什么原因。
> > 
> > 
> > 我的正好也是K2T集客固件！
> 
> 嗯， 更换成K3当AP就正常了，K2T反复改了几个无线配置，还是连不上。

我也遇到这问题了，K2T不知道为啥连不上，可是我只有K2T，我刷了openwrt也连不上。

saoog, 2020-04-18:
我也遇到了同样的问题。华硕acrh13刷的op固件，连接不上无线2.4g，换个tp路由随便连接成功。唉。这到底怎么解决？加密协议不能向下兼容？希望作者能更新，谢谢

a2633063, 2020-09-20:
> 我也遇到了同样的问题。华硕acrh13刷的op固件，连接不上无线2.4g，换个tp路由随便连接成功。唉。这到底怎么解决？加密协议不能向下兼容？希望作者能更新，谢谢

可以设置你的路由器把加密协议设置相同试试

asiacny, 2024-08-27:
有标点符号或特殊字符的就连不上，只有大小写的则正常，相同特殊字符的密码在dc1插座上正常