a2633063/zM1 #10: 请问如何修改悟空zM1的时间

birdxs, 2020-03-11:
固件版本V0.0.7
zM1的显示时间不准，手机APP没找到修改接口，请问后续固件支持修改或者对时吗？

honentan, 2020-03-18:
我也有类似问题：时钟比北京时间快25小时，看起来也不像时区设置错误。如何调整时间？

honentan, 2020-03-18:
> 固件版本V0.0.7
> zM1的显示时间不准，手机APP没找到修改接口，请问后续固件支持修改或者对时吗？
> 一开始zM1时间提前40分钟，点击手动校时后时间提前了65分钟，还是与当前时间对不上。

我的版本v0.03，可以手动更新。现升级到v0.07再看看有没有问题。

birdxs, 2020-03-20:
可能跟网络有关，宿舍电信校准不对，家里移动网络校准是对的，但是过几天就又跑偏了。

a2633063, 2020-04-14:
v0.0.8之前的版本每天零点校时   v0.0.8每天中午十二点及零点自动校时.
也可以在app的设备设置页面点手动校时按钮立刻校时

haxiaofu, 2020-04-22:
用用就快1小时  不知道什么问题

a2633063, 2020-06-23:
v0.1.3已正式发布,可ota测试是否还有此问题

1iu, 2020-09-14:
请问点击手动校时失败可能会是什么原因？

数据同步一切正常，在MQTT和UDP模式下点击手动校准都提示`校时失败，请重试`，ZControl APP首页下方日志显示`校时失败，请确认时间正确`

时间被重置到2000-1-1 00:00了

a2633063, 2020-09-14:
zM1校时使用以下服务器校时,从第一个服务器开始校时,如果失败,尝试下一个,如果全部失败则返回`校时失败，请重试`
请确认你的网络环境能够正常连接以下服务器中的一个,也可以换个时间再试试.
这些服务器都是些常用的sntp校时服务器
```
ntp1.aliyun.com
time1.cloud.tencent.com
cn.ntp.org.cn
pool.ntp.org
cn.pool.ntp.org
s1a.time.edu.cn
ntp.sjtu.edu.cn
```

1iu, 2020-09-14:
> zM1校时使用以下服务器校时,从第一个服务器开始校时,如果失败,尝试下一个,如果全部失败则返回`校时失败，请重试`
> 请确认你的网络环境能够正常连接以下服务器中的一个,也可以换个时间再试试.
> 这些服务器都是些常用的sntp校时服务器
> 
> ```
> ntp1.aliyun.com
> time1.cloud.tencent.com
> cn.ntp.org.cn
> pool.ntp.org
> cn.pool.ntp.org
> s1a.time.edu.cn
> ntp.sjtu.edu.cn
> ```

非常感谢回复，在同路由器下用电脑测试以上NTP服务器绝大部分可访问。

最后在路由器中配置自定义DNS，将ntp1.aliyun.com指向路由器IP，并在路由器上开启NTP服务。随后M1可同步时间。

具体原因未知，看起来是我的M1访问不了外网了:rofl:

closed 2020-09-21 by a2633063