a2633063/zM1 #14: 无法连接WiFi,指示灯和X呈

AlioDream, 2020-04-14:


AlioDream, 2020-04-14:
固件TTL口输出：（已从网页配网）（无法输入任何字符）
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1128:10002000]
IMG1 ENTRY[800046f:100021ef]
[233][MOC MAIN: moc_main.c: 199] Lib version 3080B002.024. APP built time 17:48:14
[243][MOC MAIN: moc_main.c: 200] heap reuse from 0x1003101c, 57300 bytes
[251][RTOS: mico_rtos_common.c:  84] Started MiCO RTOS interface for mocOS v1.0.0
[259][M1: main.c:  78] Start v0.0.8

使用HA论坛的mqtt固件输出：（无法输入任何字符）（无法配网）
ROM:[V0.1]
FLASHRATE:4
BOOT TYPE:0 XTAL:40000000
IMG1 DATA[1128:10002000]
IMG1 ENTRY[800046f:100021ef]
[233][MOC MAIN: moc_main.c: 199] Lib version 3080B002.024. APP built time 17:48:14
[243][MOC MAIN: moc_main.c: 200] heap reuse from 0x1003101c, 57300 bytes
[251][RTOS: mico_rtos_common.c:  84] Started MiCO RTOS interface for mocOS v1.0.0

a2633063, 2020-04-14:
不明白你说的变暗是什么意思...
能否拍照上传
网页输入WiFi信息后重启wifi,wifi led闪烁表示正在连接wifi
如果连接失败wifi和红色叉led都会恒亮.
wifi的led亮度和屏幕有关  若关闭屏幕 wifiled会以非常低的亮度点亮

AlioDream, 2020-04-14:
[image]

AlioDream, 2020-04-14:
配网完成后重启指示灯就会变暗。开关屏幕等都无变化.
按15s全屏幕会点亮，但WIFI指示灯和红X不会变亮，也无法重新配网.

AlioDream, 2020-04-14:
Jlink，TTL刷机后同样结果，感觉是模块损坏。
不折腾M1了，以后有时间不如做一个检测器。

closed 2020-04-14 by AlioDream