a2633063/zM1 #20: ZM1 v0.1.2 时间错误

NETUF0, 2020-06-18:
时间随机快61分钟，具体出现机率不明，手机APP上手动校时可恢复。 在ROS 设置静态域名，将 ntp1.aliyun.com 等几个NTP服务 指向ROS上自建的NTP服务，仍然会出现时间错误。 
问题无法手动重现，即使 在ROS上禁止M1 访问互联网，然后在手机APP上校时，也不会出现时间加快61分钟情况。 
未设置 MQTT 和ZA1 联动
定时亮度 
01：00关
06：00 2
11：00 4
17：00 2

NETUF0, 2020-06-19:
好像把定时亮度停了就好了，估计触发bug的原因就是定时亮度和自动校时的时间上有重叠，两个函数同时执行造成的冲突？

a2633063, 2020-06-19:
> 好像把定时亮度停了就好了，估计触发bug的原因就是定时亮度和自动校时的时间上有重叠，两个函数同时执行造成的冲突？

实际并没有  时间异常的问题概率较低
可以多一点时间看看 我也将此消息转给群内其他人帮忙试试看

ma5528374, 2020-06-20:
我也遇见了和你一样的问题，隔几天时间就会快61分钟甚至日期还比现在快了一天。发生好几次了，不是小概率事件。按照你说的我自己关闭定时亮度了看看是否能解决这个问题

NETUF0, 2020-06-20:
> 我也遇见了和你一样的问题，隔几天时间就会快61分钟甚至日期还比现在快了一天。发生好几次了，不是小概率事件。按照你说的我自己关闭定时亮度了看看是否能解决这个问题

哈哈，握爪！ 刷之前是快几分钟，然后刷的时候和老婆吹牛说，弄好了再也不会快了，结果变成快一个小时，被各种鄙视（︶︿︶）=凸

NETUF0, 2020-06-20:
> > 好像把定时亮度停了就好了，估计触发bug的原因就是定时亮度和自动校时的时间上有重叠，两个函数同时执行造成的冲突？
> 
> 实际并没有  时间异常的问题概率较低
> 可以多一点时间看看 我也将此消息转给群内其他人帮忙试试看

实际我遇到发生的概率并不低，对时后不超过两天时间就会错。 从我停掉自动亮度这几天一直正常

a2633063, 2020-06-22:
zM1时间不对   ota到0.1.3测试版本试试看
长按版本号 输入后确认:
http://zipzhang.top/zM1/v0.1.3/zM1%40MK3080B%40moc.ota.bin

此固件仅为测试时间异常的问题 可以试试

a2633063, 2020-06-23:
v0.1.3已正式发布,可ota测试是否还有此问题

NETUF0, 2020-06-25:
> v0.1.3已正式发布,可ota测试是否还有此问题

已经更新了，观察几天再来反馈！

NETUF0, 2020-06-28:
> v0.1.3已正式发布,可ota测试是否还有此问题

试了两三天，目前都正常，自动亮度也都打开了

NETUF0, 2020-07-04:
> v0.1.3已正式发布,可ota测试是否还有此问题

一周也没出过同样问题，应该是解决了

closed 2020-07-09 by a2633063