a2633063/zM1 #23: 我有个想法，可否在M1里再塞入一个二氧化碳传感器？

afirmware, 2020-09-10:
最近淘宝有些二氧化碳传感器也便宜了，如果DIY二氧化碳传感器还需要再占一个ESP8266。突然想到M1作为一个空气质量检测仪就差一个二氧化碳传感器了，如果能在M1里再塞入一个就好了。
这个主控EMW3080B是否还有空余的脚位能用于RX TX和传感器通信？如果有的话能否考虑加上这种功能？

a2633063, 2020-09-10:
有 但是我手上没有二氧化碳传感器

afirmware, 2020-09-11:
我前段时间在一家淘宝二手店买过这种，https://item.taobao.com/item.htm?id=616435137980 20元一个接上tx rx 能输出ppm，但看评论有人提到可能不准，不过我也没其他模块对比就当娱乐。如你需要我可以寄给你。

另外最近还发现另一家便宜的二氧化碳模块30元一个，https://item.taobao.com/item.htm?id=618737453641 我还没买，不知道数据什么样。

a2633063, 2020-09-20:
20元的含USB可能体积太大需要拆下，30那个可能好点
注意下，看到二氧化碳模块一般价格100元左右，价格太低不建议入手，
可以先邮件联系我私聊

afirmware, 2020-10-04:
20元的那个USB是需要拆下的，是个插针拔掉就可以，然后我试了下正好可以塞下。
给您发邮件了。

kingandwood, 2020-11-06:
搭楼同求！

nkh0472, 2022-02-13:
> 我前段时间在一家淘宝二手店买过这种，https://item.taobao.com/item.htm?id=616435137980 20元一个接上tx rx 能输出ppm，但看评论有人提到可能不准，不过我也没其他模块对比就当娱乐。如你需要我可以寄给你。
> 
> 另外最近还发现另一家便宜的二氧化碳模块30元一个，https://item.taobao.com/item.htm?id=618737453641 我还没买，不知道数据什么样。

光学二氧化碳模块的光源会随使用逐渐衰减，几年就不准了。*部分*具有定期校准功能的二氧化碳传感器需要定期把设备带电放置在户外约5分钟（可以手动通过引脚触发校准），使数据校准为大气400ppm的水平，不放在户外的话自动校准也没用。总的来说稍显鸡肋……
价格的话确实通常是100块以上的才靠谱一些。

closed 2023-07-21 by a2633063