a2633063/zM1 #28: 請問可否自訂NTP Server ?

VenceFu, 2021-02-08:
請問可否自訂NTP Server ?

a2633063, 2021-02-18:
当前使用以下的ntp服务器:
```
ntp1.aliyun.com
time1.cloud.tencent.com
cn.ntp.org.cn
pool.ntp.org
cn.pool.ntp.org
s1a.time.edu.cn
ntp.sjtu.edu.cn
```
从第一个开始校时,若校时失败则尝试第二个依次尝试.
如果需要自订NTP Server,可以直接修改路由器host劫持 `ntp1.aliyun.com`

VenceFu, 2021-02-20:
非常感謝你，這樣可以用
未來如果可以，還是希望在設定畫面製作一個欄位，讓使用者可填寫NTP Server的
謝謝

DotnetChen, 2022-01-23:
我把 ntp1.aliyun.com 脅持到  118.163.81.63
並且送了 {"mac": "bxxxxxx","time": -1 } 到設備上, 設備還是會回傳 "time": -1

以確認設備發送的 ntp1.aliyun.com request 有經過自己架設的 DNS Server 轉發
client 192.168.2.32#17201 (ntp1.aliyun.com): query: ntp1.aliyun.com IN A + (192.168.2.1)

也使用了 NTP 工具確認經由 DNS 脅持後的 ntp1.aliyun.com 是正常服務的
09:54:15.061::查詢 ntp1.aliyun.com 主機 IP...
09:54:15.064::主機 IP 為 118.163.81.63
09:54:15.065::訊息送出,等待主機回應,請稍待....
09:54:15.085::系統時間調快 0.002 秒

closed 2023-03-01 by a2633063