a2633063/zM1 #44: homeassistant更新后不支持light的mqtt配置

style6, 2022-12-09:
homeassistant更新后不支持light的mqtt配置

luosxn, 2022-12-21:
##mqtt-空气检测仪##
## 以下内容中,请将MACMAC替换为你的排插的mac地址,不带冒号,全部小写,如123456789abc
## https://github.com/a2633063/zM1/wiki/homeassistant%E6%8E%A5%E5%85%A5
##搬运于https://bbs.hassbian.com/thread-10299-1-1.html帖子的@vr7jj回复
mqtt:
  sensor:
    - name: 'zm1_b0f893224614_temperature'
      state_topic: 'device/zm1/b0f893224614/sensor'
      unique_id: zm1_b0f893224614_temperature
      unit_of_measurement: '°C'
      icon: 'mdi:thermometer'
      value_template: '{{ value_json.temperature }}'
    - name: 'zm1_b0f893224614_humidity'
      unique_id: zm1_b0f893224614_humidity
      state_topic: 'device/zm1/b0f893224614/sensor'
      unit_of_measurement: '%'
      icon: mdi:water-percent
      value_template: '{{ value_json.humidity }}'
    - name: 'zm1_b0f893224614_pm25'
      unique_id: zm1_b0f893224614_pm25
      state_topic: 'device/zm1/b0f893224614/sensor'
      unit_of_measurement: 'μg/m³'
      icon: mdi:blur
      value_template: '{{ value_json.PM25 }}'
    - name: 'zm1_b0f893224614_hcho'
      unique_id: zm1_b0f893224614_hcho
      state_topic: 'device/zm1/b0f893224614/sensor'
      unit_of_measurement: 'mg/m³'
      icon: mdi:chemical-weapon
      value_template: '{{ value_json.formaldehyde }}'

  light:
    - name: zm1_b0f893224614_brightness
      unique_id: zm1_b0f893224614_brightness
      schema: template
      command_topic: "device/zm1/b0f893224614/set"
      state_topic: "device/zm1/b0f893224614/state"
      command_on_template: >
        {"mac": "b0f893224614"
        {%- if brightness is defined -%}
        , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
        {%- else -%}
        , "brightness": 4
        {%- endif -%}
        }
      command_off_template: '{"mac": "b0f893224614", "brightness": 0}'
      state_template: >
        {%- if value_json.brightness == 0 -%}
          off
        {%- else -%}
          on
        {%- endif -%}
      brightness_template: >
        {%- if value_json.brightness is defined -%}
          {{ ( value_json.brightness *64 )|int }}
        {%- endif -%}

homeassistant:
  customize:
    light.zm1_b0f893224614_brightness:
      friendly_name: M1亮度
    sensor.zm1_b0f893224614_temperature:
      friendly_name: M1温度
    sensor.zm1_b0f893224614_humidity:
      friendly_name: M1湿度
    sensor.zm1_b0f893224614_pm25:
      friendly_name: M1 PM2.5
    sensor.zm1_b0f893224614_hcho:
      friendly_name: M1甲醛

##搬运于https://bbs.hassbian.com/thread-10299-1-1.html帖子的@vr7jj的回复

bluskai, 2023-05-19:
```
mqtt:
  sensor:
  - name: 'zm1_MACMAC_temperature'
    state_topic: 'device/zm1/MACMAC/sensor'
    unique_id: zm1_MACMAC_temperature
    unit_of_measurement: '°C'
    icon: 'mdi:thermometer'
    value_template: '{{ value_json.temperature }}'
  - name: 'zm1_MACMAC_humidity'
    unique_id: zm1_MACMAC_humidity
    state_topic: 'device/zm1/MACMAC/sensor'
    unit_of_measurement: '%'
    icon: mdi:water-percent
    value_template: '{{ value_json.humidity }}'
  - name: 'zm1_MACMAC_pm25'
    unique_id: zm1_MACMAC_pm25
    state_topic: 'device/zm1/MACMAC/sensor'
    unit_of_measurement: 'μg/m³'
    icon: mdi:blur
    value_template: '{{ value_json.PM25 }}'
  - name: 'zm1_MACMAC_hcho'
    unique_id: zm1_MACMAC_hcho
    state_topic: 'device/zm1/MACMAC/sensor'
    unit_of_measurement: 'mg/m³'
    icon: mdi:chemical-weapon
    value_template: '{{ value_json.formaldehyde }}'

  light:
  - name: zm1_MACMAC_brightness
    unique_id: zm1_MACMAC_brightness
    schema: template
    command_topic: "device/zm1/MACMAC/set"
    state_topic: "device/zm1/MACMAC/state"
    command_on_template: >
      {"mac": "MACMAC"
      {%- if brightness is defined -%}
      , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
      {%- else -%}
      , "brightness": 4
      {%- endif -%}
      }
    command_off_template: '{"mac": "MACMAC", "brightness": 0}'
    state_template: >
      {%- if value_json.brightness == 0 -%}
        off
      {%- else -%}
        on
      {%- endif -%}
    brightness_template: >
      {%- if value_json.brightness is defined -%}
        {{ ( value_json.brightness *64 )|int }}
      {%- endif -%}

homeassistant:
  customize:
    light.zm1_MACMAC_brightness:
      friendly_name: M1亮度
    sensor.zm1_MACMAC_temperature:
      friendly_name: M1温度
    sensor.zm1_MACMAC_humidity:
      friendly_name: M1湿度
    sensor.zm1_MACMAC_pm25:
      friendly_name: M1 PM2.5
    sensor.zm1_MACMAC_hcho:
      friendly_name: M1甲醛
```

a2633063, 2023-06-19:
wiki中配置已经修改

closed 2023-06-19 by a2633063