a2633063/zM1 #50: 不断地循环输出  ERROR: UNKNOWN COMMAND

John-zhan, 2023-04-27:
ERROR: UNKNOWN COMMAND

我这接上340 后一直在串口收到不断地上面的显示，怎么回事了。我这里直接用M1 的usb 供电模块了。19脚的GND 是跟340 模块的GND 连着。

John-zhan, 2023-04-27:
设备管理器的波特率没有 921600 需要怎么设置。我目前是9600

a2633063, 2023-05-12:
> 设备管理器的波特率没有 921600 需要怎么设置。我目前是9600

和设备管理器无关,使用串口软件打开时设置波特率即可

a2633063, 2023-05-12:
> ERROR: UNKNOWN COMMAND
> 
> 我这接上340 后一直在串口收到不断地上面的显示，怎么回事了。我这里直接用M1 的usb 供电模块了。19脚的GND 是跟340 模块的GND 连着。

wifi模组与mcu主控也通过此串口通信 因此若正好此时有通信 此处就会有乱码 这也是部分刷机时刷一部分失败的原因
另外也可能是你接线有问题/接触不良等问题 需要你自行排除问题
建议使用jlink刷成功率会高很多

closed 2023-07-21 by a2633063

John-zhan, 2023-12-10:
> > 设备管理器的波特率没有 921600 需要怎么设置。我目前是9600
> 
> 和设备管理器无关,使用串口软件打开时设置波特率即可

今天刚看了官网的设备文档，使用的不是CH340 这个usb转ttl 工具，用340 工具设置波特率也是921600吗？设备管理器的com串口属性里，ch340 最高的波特率是128000。
我这里进bootloader 后是这样子的，接触不良可以排除，这次我直接用焊锡把线焊接到接点了。无线模块通信干扰怎么能关掉呢，不然进入bootloader 操作啊
[image]

John-zhan, 2023-12-10:
村里的电源没有地线，会导致什么问题吗

casojie, 2025-06-20:
我也同样出现一样的问题，结果换另外一个usb转ttl就ok了