a2633063/zM1 #51: 请教多台M1设备连接HA的方法。

luz83o, 2023-07-17:
发现以一下代码添加多台后有错误提示。MAC地址已经修改，但还是一直提示名字冲突。

第一台代码：
  light:
      name: zm1_b0f89326ff10_brightness
      unique_id: zm1_b0f89326ff10_brightness
      schema: template
      command_topic: "device/zm1/b0f89326ff10/set"
      state_topic: "device/zm1/b0f89326ff10/state"
      command_on_template: >
        {"mac": "b0f89326ff10"
        {%- if brightness is defined -%}
        , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
        {%- else -%}
        , "brightness": 4
        {%- endif -%}
        }
      command_off_template: '{"mac": "b0f89326ff10", "brightness": 0}'
      state_template: >
        {%- if value_json.brightness == 0 -%}
          off
        {%- else -%}
          on
        {%- endif -%}
      brightness_template: >
        {%- if value_json.brightness is defined -%}
          {{ ( value_json.brightness *64 )|int }}
        {%- endif -%}

第二台代码：
  light:
      name: zm1_b0f89323adaa_brightness
      unique_id: zm1_b0f89323adaa_brightness
      schema: template
      command_topic: "device/zm1/b0f89323adaa/set"
      state_topic: "device/zm1/b0f89323adaa/state"
      command_on_template: >
        {"mac": "b0f89323adaa"
        {%- if brightness is defined -%}
        , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
        {%- else -%}
        , "brightness": 4
        {%- endif -%}
        }
      command_off_template: '{"mac": "b0f89323adaa", "brightness": 0}'
      state_template: >
        {%- if value_json.brightness == 0 -%}
          off
        {%- else -%}
          on
        {%- endif -%}
      brightness_template: >
        {%- if value_json.brightness is defined -%}
          {{ ( value_json.brightness *64 )|int }}
        {%- endif -%}

请教还需要修改哪里？

a2633063, 2023-07-21:
请贴错误提示,正常不会有重复的提示

luz83o, 2023-07-22:
> 请贴错误提示,正常不会有重复的提示

感谢你的回复，已经发现问题了。有同样问题的朋友可以参考下面的代码。

第一台代码：
- name: zm1_b0f89326ff10_brightness
unique_id: zm1_b0f89326ff10_brightness
第二台代码：
- name: zm1_b0f89323adaa_brightness
unique_id: zm1_b0f89323adaa_brightness

Darkcat222151, 2024-06-22:
light:
#第一台设备
      name: zm1_b0f89321a1d6_brightness
      unique_id: zm1_b0f89321a1d6_brightness
      schema: template
      command_topic: "device/zm1/b0f89321a1d6/set"
      state_topic: "device/zm1/b0f89321a1d6/state"
      command_on_template: >
        {"mac": "b0f89321a1d6"
        {%- if brightness is defined -%}
        , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
        {%- else -%}
        , "brightness": 4
        {%- endif -%}
        }
      command_off_template: '{"mac": "b0f89321a1d6", "brightness": 0}'
      state_template: >
        {%- if value_json.brightness == 0 -%}
          off
        {%- else -%}
          on
        {%- endif -%}
      brightness_template: >
        {%- if value_json.brightness is defined -%}
          {{ ( value_json.brightness *64 )|int }}
        {%- endif -%}
#第二台设备
      name: zm1_b0f8931ea081_brightness
      unique_id: zm1_b0f8931ea081_brightness
      schema: template
      command_topic: "device/zm1/b0f8931ea081/set"
      state_topic: "device/zm1/b0f8931ea081/state"
      command_on_template: >
        {"mac": "b0f8931ea081"
        {%- if brightness is defined -%}
        , "brightness": {{ ((brightness-1) / 64 )|int +1 }}
        {%- else -%}
        , "brightness": 4
        {%- endif -%}
        }
      command_off_template: '{"mac": "b0f8931ea081", "brightness": 0}'
      state_template: >
        {%- if value_json.brightness == 0 -%}
          off
        {%- else -%}
          on
        {%- endif -%}
      brightness_template: >
        {%- if value_json.brightness is defined -%}
          {{ ( value_json.brightness *64 )|int }}
        {%- endif -%}
[image]
 请问您知道为什么多台zmi的设备显示控制和亮度开关只能最后一个有效呢？甲醛什么的都拾取正常。

Darkcat222151, 2024-06-28:
问题已经解决，多个light
命令应如下格式
light：
  - name：XXXXX
     ...
     #已有第一个是设备的教程的代码
  - name：XXXXX
      ...
     #已有第二个是设备的教程的代码 
#后续以此类推。需要注意格式

Darkcat222151, 2024-06-28:
这里小横杠打不出来，变成点了。。。。