a2633063/zM1 #54: 关于通信协议中的部分数据格式类型问题

Latezly, 2024-02-26:
在`device/zm1/MAC/sensor``中
```
{
	"mac":"d0bae4618552",	//唯一标识符
	"PM25":"1.0",				//当前PM2.5
	"formaldehyde":"0.2",		//甲醛
	"temperature":"20",			//温度
	"humidity":"64",			//湿度
}
```
除了mac以外，PM2.5、甲醛、温度、湿度的数据实际上是可以用int或者float类型来表达的，带入到json中只需要删掉value的双引号即可
```
mac                 string
PM25                int
formaldehyde        float
temperature         float
humidity            float
```

我在用telegraf通过mqtt抓取数据并使用json_v2解析并写入influxdb的过程中，json_v2会自动将M2.5、甲醛、温度、湿度以string的格式写入到measurement里面
```
> show field keys from phicomm
name: phicomm
fieldKey            fieldType
--------            ---------
mac                 string
PM25                string
formaldehyde        string
temperature         string
humidity            string
```

对于这些问题不知作者觉得是否可以改善一下？

a2633063, 2024-02-26:
最开始是考虑将单位加上,所以做成了string   之后未加单位  但是部分人已经使用string来处理了  所以一直未更改为int或float
确实改为int或float更为合理   但考虑大部分已经使用string处理数据  改为float后大部分人要改   此处先暂时不改

Latezly, 2024-02-26:
那能否辛苦作者做一个预发布版本呢？作为一个可选项之类的。因为闭源的原因我想自行修改也没办法